SQ9MDD/TTGO-T-Beam-LoRa-APRS #29: Feature Request: iGate Integrated in the Firmware

eaxi, 2021-04-14:
Integration of the traditional iGate Feature from Peterus.

So that we only need one Firmware for the Boards.

Config for the iGates from the Web-Interface.

M0IGA, 2021-06-11:
simple digi rather than iGate

mi-gri, 2021-08-30:
I second the request. I would like to have the **option** to send the received packets directly to the aprsis server while using KISS to send my own packets (e.g. using an app on my mobile phone) . Using KISS (especially for both directions) does end in a significant lag comparing to a direct aprsis connection. That way the station is nearly never the first to announce a packet if there is any other receiving station. So it seems to have a very bad receiving according to the aprs-website although the packets are just transferred to late. It also causes a big amount of dupes on the apris-server according the statistics.

mattbk, 2021-09-20:
Would integrating pieces of the LoRa_APRS_iGate project be useful? https://github.com/lora-aprs/LoRa_APRS_iGate

As an aside, would packets sent from a device using the code in the current repository be picked up by the LoRa_APRS_iGate project (assuming same frequency, which they seem to hardcode)? I don't yet understand how the packets are being sent via LoRa.

mattbk, 2021-09-23:
I can confirm that the https://github.com/lora-aprs/LoRa_APRS_iGate project can act as an iGate for a device running this code, if the baud rate is set at 300, not 1200. SQ9MDD's code can both send and receive packets sent by the lora-aprs code, and vice versa.

This doesn't solve the ticket, but does make it possible to deploy an iGate/digipeater if you are OK using different code for it.

Some notes I took while playing around.

SQ9MDD:
[image]
[image]
These codes mean:
- 300 baud: `Bw125Cr45Sf4096`
	- bandwidth 125 kHz
	- coding rate 4/5
	- spreading factor 4096 chips/symbol
- 1200 baud: `Bw125Cr47Sf512`
	- bandwidth 125 kHz
	- coding rate 4/7
	- spreading factor 512 chips/symbol

lora-aprs:
[image]

Spreading factor is here given in different units? A [number](https://www.thethingsnetwork.org/docs/lorawan/spreading-factors/) between 7 and 12.

2^{𝑆𝐹} 𝑐ℎ𝑖𝑝𝑠 =1 𝑠𝑦𝑚𝑏𝑜𝑙. So a spreading factor of 12 is 4096 chips/symbol.

iu2frl, 2021-09-27:
i have the iGate software you mentioned and it's working just fine only by setting baudrate to 300 in the TTGO Tracker, we already have two iGate in my town and the coverage it's quite impressive considering the low power

dl9sau, 2021-12-18:
I'll try to implement that feature in the next weeks.

dl9sau, 2022-01-21:
If you like to test my brandnew version, please send my a mail at my callsing at darc.de

I'd appreciate feedback if everything works as expected and proves stable.