SQ9MDD/TTGO-T-Beam-LoRa-APRS #40: What is the default WIFI (N0CALL-0 AP) password?

Ge0Lin, 2021-07-26:
Hi
I flashed my T-BEAM, but it doesn’t seem to boot properly.

Leaving...
Hard Resetting...
Done! Flashing is complete!

Showing logs:
[16:52:43][E][Preferences.cpp:437] getString(): nvs_get_str len fail: aprs_s_table NOT_FOUND
[16:52:43][E][Preferences.cpp:437] getString(): nvs_get_str len fail: aprs_symbol NOT_FOUND
[16:52:48][E][Preferences.cpp:437] getString(): nvs_get_str len fail: aprs_callsign NOT_FOUND
[16:52:49][E][Preferences.cpp:437] getString(): nvs_get_str len fail: wifi_password NOT_FOUND
[16:52:49][E][Preferences.cpp:437] getString(): nvs_get_str len fail: wifi_ssid NOT_FOUND

73
de BG0GE

SQ9MDD, 2021-07-26:
pasword is 10 mal x

xxxxxxxxxx

enjoy.

R.

Ge0Lin, 2021-07-27:
TNX!
73

SQ9MDD, 2021-07-30:
I have added the information at the end of the readme.md I am closing the ticket

closed 2021-07-30 by SQ9MDD