SQ9MDD/TTGO-T-Beam-LoRa-APRS #41: LoRa APRS in Packet mode

Excalibur201010, 2021-07-31:
Good evening.
1. I want to thank you for your work in the development of this tracker
2. At the moment I am the only one in Ukraine who started using and developing the APRS network based on LoRa and I have a question for you.
3. I connected my LoRa module to BPQ32 software, DIGI / IGATE mode works fine, but I would also like to enable packet mode so that others can connect to WinLink network via LoRa and my Node.So when I send a command to connect from another device (WoAD + LoRa module and your firmware), my node does not respond, can this be somehow implemented in this firmware? If I understand correctly, then the SABM command does not pass. Maybe i am doing something wrong?
4. I also noticed that any program for working with TNC(Agwpe Packet engine) sends commands TxDelay, Textail, Persistence and all this goes on the air, although it should not.
[image]

Sorry for my English and Thanks. Best regards US1GHQ - Alexsandr 73!

Qyon, 2021-08-21:
TNC sending control data error fixed in PR #45 But this PR also provides many other changes, that need to be tested (mainly in TNC via Wifi handling code).
General use TNC (suitable for winlink etc.) is in my plans, but some details need more thought.

Excalibur201010, 2021-08-21:
Very big thanks, i am trying testing tomorrow.
My TNC connected to bpq32 over wifi, so i am testing all features.
If needed some information i am ready help 
Best regards and 73 - US1GHQ Alexsandr

SQ9MDD, 2021-08-22:
Corrections are available:
https://github.com/SQ9MDD/TTGO-T-Beam-LoRa-APRS/releases/tag/refs%2Fheads%2Fmaster-2021-08-22T071803

I am waiting for confirmation that the changes are working.

US1GHQ, 2021-08-22:
Good evening all.
So i am tested new changes.
1. Then programm (AGWPE or bpq32) send commands tx tail, persiserence, and other commands all fine
2. Other commands for connecting to bbs winlink As said Qyon need time to be implemented
3. I found bug with Web interface after all changes, not showing what is received
[image]

So maybe do not close this issue
Other works is fine.
Many big thanks =)
And sorry for my English.

Qyon, 2021-08-22:
> I found bug with Web interface after all changes, not showing what is received

Only APRS packets will be shown here.

// 73 SQ5RWU

US1GHQ, 2021-08-22:
I know, After the update, it does not show anything at all.
My second TNC is running previous version of firmware and showing all aprs packets =)
[image]

US1GHQ, 2021-09-06:
Good evening all, i am try reaize PACKET MODE and there is little progress.
Now i can connect to the WINLINK/BBS node but i cannot send my current position over LoRa APRS, because compatibility is broken.
now I am thinking how to solve this issue so that the TNC automatically switches between APRS and PACKET mode.
then i finished i create the pull request.
[image]

US1GHQ, 2021-09-13:
Hello all.
After many days I am implement PACKET MODE, now you can receive and send mail via BBS, if the appropriate software is installed on the node.
The implementation is far from ideal, but it works stable for me.
Now the stage of intensive testing is beginning, and, if possible, I will optimize and fix bugs if found.
Also retained compatibility with the current APRS network mode.
I do not create a pull request yet, until I am sure that everything works well.
Those interested can try my fork here. https://github.com/US1GHQ/TTGO-T-Beam-LoRa-APRS.git
[image]

US1GHQ, 2021-09-20:
Hi all
After long time i realize Packet mode for using with WINLINK/BBS, since I'm a beginner in programming,maybe someone can do better
The main idea is , by default TNC works in PACKET mode, as soon as the packet contains the word (WIDE, RELAY, ECHO) TNC switches to plain text mode (OE MODE) processes the packet and goes back to PACKET MODE.
All the changes that I made are in my personal GIT repository. It also remains to implement the CSMA (CAD) mode, as without it, very interesting results are obtained ;-)
http://git.mis.ks.ua/US1GHQ/TTGO-T-Beam-LoRa-APRS.git
Short video demonstrating receive Winlink message.
https://youtu.be/XouUY27LVts