SQ9MDD/TTGO-T-Beam-LoRa-APRS #44: change Tx frequency but cannot save

VR2XKP, 2021-08-07:
Thank you for your work !

But while I try to save the frequency (435.775), it cannot save. If I change back to 433.775, it can save and return to the top of the web page.

I am using an Android 11 phone. I had captured the screen for reference:

https://drive.google.com/file/d/188YR-YYDwevUyLWfUb6W4nZayHQNoxBC/view?usp=sharing

Thank you
VR2XKP

_Originally posted by @SQ9MDD in https://github.com/SQ9MDD/TTGO-T-Beam-LoRa-APRS/issues/35#issuecomment-892171425_

Excalibur201010, 2021-08-07:
Good evening
Because its limited in source code, if you want change freq to 435Mhz and higher please edit index.html and find this line
<input name="lora_freq" id="lora_freq" type="number" min="433.000" max="434.000" step="0.001" title="LoRa center frequency between 433.001 and 434.000">
and set "max" value to the needed freq
for example 
<input name="lora_freq" id="lora_freq" type="number" min="433.000" max="440.000" step="0.001" title="LoRa center frequency between 433.001 and 440.000">
then rebuild from source

SQ9MDD, 2021-08-21:
what is the maximum according to your bandplan? I will change it globally for everyone.

US1GHQ, 2021-08-21:
I my country i can work from 430 to 440MHz ;-)

VR2XKP, 2021-08-21:
440MHz, thank you.

silasmoeckel, 2021-08-22:
I'll add US goes to 450mhz

SQ9MDD, 2021-08-22:
Fixed.

https://github.com/SQ9MDD/TTGO-T-Beam-LoRa-APRS/releases/tag/refs%2Fheads%2Fmaster-2021-08-22T071803

closed 2021-08-22 by SQ9MDD

mattbk, 2021-09-01:
USA goes up to 928 MHz: https://www.rfwireless-world.com/Tutorials/LoRa-channels-list.html