SQ9MDD/TTGO-T-Beam-LoRa-APRS #47: Display not working with Lora32 v2.1 version

mi-gri, 2021-08-30:
Using the version for my Lora32 v2.1 board and the display keeps dark. I tried the Lora32 v2 version of the software instead and the display works.

US1GHQ, 2021-08-30:
If you build from source, find this code in TTGO_T-Beam_LoRa_APRS.ino
`#elif LORA32_21
  #define I2C_SDA 4
  #define I2C_SCL 15
  #define BUTTON 2                  //pin number for BUTTO
  #define BUZZER 13                 // enter your buzzer pin gpio
  const byte TXLED  = 4;            //pin number for LED on TX Tracker
`
and change to
`#elif LORA32_21
  #define I2C_SDA 21
  #define I2C_SCL 22
  #define BUTTON 2                  //pin number for BUTTO
  #define BUZZER 13                 // enter your buzzer pin gpio
  const byte TXLED  = 4;            //pin number for LED on TX Tracker
`

mi-gri, 2021-08-31:
Thank you - that worked for me.

iu2frl, 2021-09-15:
is this a custom board or should we make it definitive?

mi-gri, 2021-09-18:
I'm not sure if this is the default for the 2.1 versions, but I have now three boards, all need this change.

iu2frl, 2021-09-19:
merged, we can close it

closed 2021-09-19 by mi-gri