SQ9MDD/TTGO-T-Beam-LoRa-APRS #48: Feature Request: integrated digipeater

mattbk, 2021-08-30:
Pulling this out as a separate ticket from https://github.com/SQ9MDD/TTGO-T-Beam-LoRa-APRS/issues/29#issuecomment-859360187:
> simple digi rather than iGate

This would make it easier to deploy a temporary digipeater without having to use a separate dedicated device.

mattbk, 2021-09-23:
If you are OK using different software for a digipeater on a dedicated device, the lora-aprs digipeater will work: https://github.com/lora-aprs/LoRa_APRS_iGate.

I added some interoperability notes at https://github.com/SQ9MDD/TTGO-T-Beam-LoRa-APRS/issues/29#issuecomment-926018368.

dl9sau, 2021-12-18:
For others searching for it or thinking about implementing it:
I have implemented and tested this feature a few weeks ago.
You can run it as WIDE1 (fill-in) digi, or as WIDE2 digi. Path reduction according to our LoRa RFF policy.
You can digipeat on the same frequency, or to a secondary frequency (with higher bitrate - useful for a fill-in-digi without internet access), or to both.
Stay tuned ;)