SQ9MDD/TTGO-T-Beam-LoRa-APRS #52: APLSxx tocall is not assigned to LoRa

ge0rg, 2021-09-03:
Hi,

your tracker code is using the hard-coded to-call of `APLS01` which is [officially assigned to sarimesh](http://www.aprs.org/aprs11/tocalls.txt). I'd suggest using the `APLT` (lora tracker) prefix with a different version number, or get a new tocall prefix assigned by Bob Bruninga. That should be as easy as sending him an email :)

Georg DO1GL

SQ9MDD, 2021-09-04:
I can't take APLT - APLTxx  LoRa Tracker - OE5BPA
We need to talk to Bob again because I see my request has been ignored:
24 Aug 21 Added APLSxx SARIMESH http://www.sarimesh.net

SQ9MDD, 2021-09-04:
http://www.aprs.org/aprs11/tocalls.txt

Bob accepted the proposal and from now on we have APLOxx address for this project

closed 2021-09-04 by SQ9MDD