SQ9MDD/TTGO-T-Beam-LoRa-APRS #59: Feature request: Use APRS telemetry for battery voltage

mattbk, 2021-09-05:
Low priority. I'm willing to help.

Battery voltage seems like it would fit better in a telemetry field than in the comment. I can also understand why this is the easiest place to put it, as some (many?) KISS clients don't show telemetry well.

SQ9MDD, 2021-09-06:
Did you see this:

[image]

mattbk, 2021-09-06:
Interesting, thank you!

mattbk, 2021-09-11:
These are mostly notes for my own benefit as I work through this.

I have tried turning on telemetry (at 1-minute interval for testing) using the following in `platformio.ini` when building:
```
-D 'ENABLE_TNC_SELF_TELEMETRY'		; send telemetry data about device
-D 'TNC_SELF_TELEMETRY_INTERVAL=60000L' ; telemetry interval (milliseconds)
```
It looks like the telemetry is being sent (in the PlatformIO monitor), but I should probably learn how to show a more detailed debug output.

My current setup is 
- one t-beam sending telemetry via Bluetooth Serial KISS, accessed by QTH software on macOS
- QTH acting as digipeater on 2M (via another KISS connection to Dire Wolf)
- local receive-only iGate receiving packets and gating them to Internet
- seeing position packets on aprs.fi and aprsdirect.com

At some point in this long chain, the telemetry is being lost, but I am not sure whether it is a bad format out of this software or out of QTH (or even the iGate). I need to shorten the chain somehow. #29 might do that.

This is the sort of thing I see in QTH:
```
09:51:44.643 DEBUG APRSManagerImpl.onAprsPacketReceived():167 - MESSAGE > from: KKKKKK-4 to: KKKKKK-4 msg: UNIT.mV,mA,mA,mV,mA
09:51:44.685 DEBUG APRSManagerImpl.onAprsPacketReceived():167 - MESSAGE > from: KKKKKK-4 to: KKKKKK-4 msg: EQNS.0,5.1,3000,0,10,0,0,10,0,0,28,3000,0,10,0
09:51:44.688 DEBUG APRSManagerImpl.onAprsPacketReceived():167 - TELEMETRY > KKKKKK-4 @ 0°0'0" N, 0°0'0" E 0
09:51:44.692 DEBUG APRSManagerImpl.onAprsPacketReceived():171 - Telemetry: 191.0,0.0,12.0,149.0,0.0,0.0
```
I don't know why telemetry position shown as 0, 0, because GPS is functioning and position packets are being sent correctly.

SQ9MDD, 2021-09-20:
wel done!

closed 2021-09-20 by SQ9MDD