SQ9MDD/TTGO-T-Beam-LoRa-APRS #66: Add OLED timeout when display is ON and no packets are received

iu2frl, 2021-09-13:
To better save power it would be useful to enable OLED just when necessary (such as when receiving packets or pressing a button). For example with OLED in power save mode one short press of user button can wake up the display and with OLED ON a short press can send manual position

iu2frl, 2021-09-13:
i got it with very few lines:

declare:
```
ulong oled_timeout = 20000;
bool tempOled = true;
ulong oled_timer = millis();
```

then create:
```
void enableOled() {
  tempOled = true;
  oled_timer = millis() + oled_timeout;
}
```

change first button detect (lines 850) to:
```
 if(digitalRead(BUTTON)==LOW && key_up == true){
    key_up = false;
    delay(50);
    if(digitalRead(BUTTON)==LOW){
      delay(300);
      time_delay = millis() + 1500;
      if(digitalRead(BUTTON)==HIGH){
        // ------- begin modification ----------
        // if OLED is off turn it on, if it is ON then send beacon manually
        if (!tempOled) {
        // turn ON OLED temporary
        enableOled();
        //----------- end -------------
        } else {
          if(gps_state == true && gps.location.isValid()){
              writedisplaytext("((MAN TX))","","","","","");
              sendpacket();
          }else{
              writedisplaytext("((FIX TX))","","","","","");
              sendpacket();
          }
        }
        key_up = true;
      }
    }
  }
```

and add a function to turn off the oled right in the main loop, right after the first button detect:
```
  // Only wake up OLED when necessary
  display.dim(!tempOled);

  if (tempOled && millis()>= oled_timer) {
    tempOled = false;
  }
  //------------------------------------
```

whenever need in the code to show anything on the OLED just run:
`enableOled()`

and that's it, more power save!

SQ9MDD, 2021-09-13:
Luca can you do a pull request from this?

iu2frl, 2021-09-14:
I'm not really into GitHub functions, i just started using it, you can find it in my fork: https://github.com/iu2frl/TTGO-T-Beam-LoRa-APRS

iu2frl, 2021-09-14:
got it! pull done

iu2frl, 2021-09-14:
Done, pull request has been sent

SQ9MDD, 2021-09-14:
I have already accepted your pull request. I have released the compilation on github,added you to the list of developers....

SQ9MDD, 2021-09-15:
After the first startup, the parameter "Display show RX time (s)" is ignored OLED is blanked right after the startup.
I wonder if this function should not have an off switch on the web page

iu2frl, 2021-09-15:
I'll get this

iu2frl, 2021-09-15:
Done, will merge it soon

i'm currently adding the HTML function

iu2frl, 2021-09-15:
I'm working on getting the parameters stored but i can't sort it out, i've added the textbox in the html page, data is being sent and saved but at next reboot it restores old value

iu2frl, 2021-09-15:
Done, added in pull request #73. Fixed first awake timer and added in html page

closed 2021-09-15 by iu2frl