SQ9MDD/TTGO-T-Beam-LoRa-APRS #81: Feedback on battery duration

iu2frl, 2021-09-19:
I'm trying to undestand the current battery consumption, the most battery aggressive feature is transmission followed by WiFi, my T-Beam lasted around 2h in tracking mode which is not so good. Any feedback?

```
2021-09-19 19:34:32 CEST: IU2FRL-12>APLO01,WIDE1-1,qAR,IU2FRL-10:!/7](AQ5&W[!0H/A=000000https://www.iu2frl.it/tracker-lora32-in-uhf/ Batt=4.04V
2021-09-19 19:34:42 CEST: IU2FRL-12>APLO01,WIDE1-1,qAR,IU2FRL-10:!/7](>Q5&N[g5H/A=000010https://www.iu2frl.it/tracker-lora32-in-uhf/ Batt=4.04V
2021-09-19 19:36:19 CEST: IU2FRL-12>APLO01,WIDE1-1,qAR,IU2FRL-10:!/7](sQ5&[[R8H/A=000008https://www.iu2frl.it/tracker-lora32-in-uhf/ Batt=4.04V
2021-09-19 19:36:36 CEST: IU2FRL-12>APLO01,WIDE1-1,qAR,IU2FRL-10:!/7](TQ5&Z[e3H/A=000004https://www.iu2frl.it/tracker-lora32-in-uhf/ Batt=4.04V
2021-09-19 19:37:46 CEST: IU2FRL-12>APLO01,WIDE1-1,qAR,IU2FRL-10:!/7](.Q5&K[I:H/A=000-10https://www.iu2frl.it/tracker-lora32-in-uhf/ Batt=4.03V
2021-09-19 19:42:43 CEST: IU2FRL-12>APLO01,WIDE1-1,qAR,IU2FRL-10:!/7](IQ5&l[I)H/A=000000https://www.iu2frl.it/tracker-lora32-in-uhf/ Batt=4.04V
2021-09-19 19:45:40 CEST: IU2FRL-12>APLO01,WIDE1-1,qAR,IU2FRL-10:!/7]&DQ5&4["AH/A=000007https://www.iu2frl.it/tracker-lora32-in-uhf/ Batt=4.03V
2021-09-19 19:50:25 CEST: IU2FRL-12>APLO01,WIDE1-1,qAR,IU2FRL-10:!/7](>Q5%)[d<H/A=0000-4https://www.iu2frl.it/tracker-lora32-in-uhf/ Batt=4.04V
2021-09-19 19:58:11 CEST: IU2FRL-12>APLO01,WIDE1-1,qAR,IU2FRL-10:!/7](DQ5&"[f&H/A=000000https://www.iu2frl.it/tracker-lora32-in-uhf/ Batt=4.03V
2021-09-19 20:15:26 CEST: IU2FRL-12>APLO01,WIDE1-1,qAR,IU2FRL-10:!/7](FQ5%l[f1H/A=0000-1https://www.iu2frl.it/tracker-lora32-in-uhf/ Batt=3.98V
2021-09-19 20:15:52 CEST: IU2FRL-12>APLO01,WIDE1-1,qAR,IU2FRL-10:!/7](_Q5&@[::H/A=000030https://www.iu2frl.it/tracker-lora32-in-uhf/ Batt=3.97V
2021-09-19 20:17:31 CEST: IU2FRL-12>APLO01,WIDE1-1,qAR,IU2FRL-10:!/7](,Q5%Y[h0H/A=000000https://www.iu2frl.it/tracker-lora32-in-uhf/ Batt=3.97V
2021-09-19 20:18:11 CEST: IU2FRL-12>APLO01,WIDE1-1,qAR,IU2FRL-10:!/7](/Q5%y[-3H/A=000027https://www.iu2frl.it/tracker-lora32-in-uhf/ Batt=3.95V
2021-09-19 20:18:20 CEST: IU2FRL-12>APLO01,WIDE1-1,qAR,IU2FRL-10:!/7]'+Q5#4[hJH/A=000017https://www.iu2frl.it/tracker-lora32-in-uhf/ Batt=3.97V
2021-09-19 20:20:30 CEST: IU2FRL-12>APLO01,WIDE1-1,qAR,IU2FRL-10:!/7]'YQ5$F[%2H/A=0000-1https://www.iu2frl.it/tracker-lora32-in-uhf/ Batt=3.97V
2021-09-19 20:20:58 CEST: IU2FRL-12>APLO01,WIDE1-1,qAR,IU2FRL-10:!/7]("Q5%R[;1H/A=000018https://www.iu2frl.it/tracker-lora32-in-uhf/ Batt=3.97V
2021-09-19 20:23:21 CEST: IU2FRL-12>APLO01,WIDE1-1,qAR,IU2FRL-10:!/7](.Q5%p[X3H/A=000001https://www.iu2frl.it/tracker-lora32-in-uhf/ Batt=3.93V
2021-09-19 20:45:13 CEST: IU2FRL-12>APLO01,WIDE1-1,qAR,IU2FRL-10:!/7](0Q5&![L5H/A=000000https://www.iu2frl.it/tracker-lora32-in-uhf/ Batt=3.88V
2021-09-19 20:45:45 CEST: IU2FRL-12>APLO01,WIDE1-1,qAR,IU2FRL-10:!/7](.Q5&U[Z6H/A=000012https://www.iu2frl.it/tracker-lora32-in-uhf/ Batt=3.88V
2021-09-19 20:50:43 CEST: IU2FRL-12>APLO01,WIDE1-1,qAR,IU2FRL-10:!/7](-Q5&K[Z+H/A=0000-9https://www.iu2frl.it/tracker-lora32-in-uhf/ Batt=3.85V
2021-09-19 20:55:45 CEST: IU2FRL-12>APLO01,WIDE1-1,qAR,IU2FRL-10:!/7](/Q5&d[5*H/A=000008https://www.iu2frl.it/tracker-lora32-in-uhf/ Batt=3.84V
2021-09-19 20:56:17 CEST: IU2FRL-12>APLO01,WIDE1-1,qAR,IU2FRL-10:!/7](<Q5'j[L<H/A=000022https://www.iu2frl.it/tracker-lora32-in-uhf/ Batt=3.83V
2021-09-19 20:57:20 CEST: IU2FRL-12>APLO01,WIDE1-1,qAR,IU2FRL-10:!/7](/Q5)1[_;H/A=000038https://www.iu2frl.it/tracker-lora32-in-uhf/ Batt=3.83V
2021-09-19 21:02:39 CEST: IU2FRL-12>APLO01,WIDE1-1,qAR,IU2FRL-10:!/7]'gQ5&?[_+H/A=000001https://www.iu2frl.it/tracker-lora32-in-uhf/ Batt=3.78V
2021-09-19 21:07:37 CEST: IU2FRL-12>APLO01,WIDE1-1,qAR,IU2FRL-10:!/7](<Q5&k[_)H/A=0000-1https://www.iu2frl.it/tracker-lora32-in-uhf/ Batt=3.72V
2021-09-19 21:12:41 CEST: IU2FRL-12>APLO01,WIDE1-1,qAR,IU2FRL-10:!/7](BQ5&M[y)H/A=000004https://www.iu2frl.it/tracker-lora32-in-uhf/ Batt=3.63V
2021-09-19 21:12:52 CEST: IU2FRL-12>APLO01,WIDE1-1,qAR,IU2FRL-10:!/7]&VQ5&][$HH/A=000005https://www.iu2frl.it/tracker-lora32-in-uhf/ Batt=3.62V
2021-09-19 21:14:51 CEST: IU2FRL-12>APLO01,WIDE1-1,qAR,IU2FRL-10:!/7])IQ5&j[I:H/A=000001https://www.iu2frl.it/tracker-lora32-in-uhf/ Batt=3.55V
2021-09-19 21:19:48 CEST: IU2FRL-12>APLO01,WIDE1-1,qAR,IU2FRL-10:!/7](MQ5&E[I*H/A=0000-5https://www.iu2frl.it/tracker-lora32-in-uhf/ Batt=3.39V
2021-09-19 21:19:59 CEST: IU2FRL-12>APLO01,WIDE1-1,qAR,IU2FRL-10:!/7]#8Q5&Q[!RH/A=000040https://www.iu2frl.it/tracker-lora32-in-uhf/ Batt=3.36V
```

mattbk, 2021-09-19:
How big is your battery in mAh?

iu2frl, 2021-09-20:
Should be around 2500mAh but I'm thinking is about to die

SQ9MDD, 2021-09-20:
definitely this battery is ruined. I will do a test today. I'll charge the battery to full and after 10hrs of use I'll put it on a charger with a mA meter and see how much is missing. I will repeat the test a few times

SQ9MDD, 2021-09-24:
Test conditions:
Sending frame every 180s, fixed beacon, GPS disabled, battery full loaded.

2021.09.23 22:45 - starting test, battery 4.18V (the battery has been charged to its maximum capacity with externall charger)
2021.09.24 07:45 - stopping test, battery 3.94V

After charging the battery back to 100% the charger showed 646mAh
So, during nine hours of operation, the device consumed 646mAh. 
That is, the power consumption is about 72mA

iu2frl, 2021-09-27:
That's not bad, can you try with GPS ON?