SQ9MDD/TTGO-T-Beam-LoRa-APRS #82: Blue blinking LED after Auto Power Off

mi-gri, 2021-09-19:
Since one of the last updates, a blue LED flashes on my T-Beam 1.1 after Auto Shut Down. 
Is this supposed to be like that?

mattbk, 2021-09-19:
Is it shut down, or just the screen turned off? A recent update turns the screen off after several seconds, and switched back on again with button press.

mi-gri, 2021-09-20:
It is shut down following the auto shut down timer: It is not connectable and sends nothing out.

SQ9MDD, 2021-09-20:
I think just before the command:
axp.shutdown();
still needs to be done:
axp.setChgLEDMode(AXP20X_LED_OFF);

SQ9MDD, 2021-09-20:
I set the tracker to turn off automatically if the voltage is less than 3.3V while turning off 4Hz signaling. Let's see if this helps

mi-gri, 2021-09-20:
Seems to work. Just added a pull request for the second shutdown code in taskWebServer.cpp

SQ9MDD, 2021-09-20:
I hope this helped. I will close the ticket if there is a problem, we will reopen it

closed 2021-09-20 by SQ9MDD