SQ9MDD/TTGO-T-Beam-LoRa-APRS #96: Cannot download any .bin file ?

miodzicho, 2021-12-22:
Cannot download any .bin file ? Getting 404

Lorafan, 2021-12-27:
same issue here also :-(
Strange, in the past the download of BIN's worked flawlessly.
Any idea what could be the reason for that issue?

SQ9MDD, 2021-12-30:
Github repo is broken try this url:

http://hamspirit.pl/SQ9MDD/TTGO-T-Beam-LoRa-APRS/

zanshee, 2022-01-20:
> Github repo is broken try this url:
> 
> http://hamspirit.pl/SQ9MDD/TTGO-T-Beam-LoRa-APRS/

Sorry, but "DOMAIN NAME: hamspirit.pl 's billing period had finished". It happened 18 jan, as I can see. Can you prolong this domain name or migrate to another and give here new link?

M0IGA, 2022-01-20:
> > Github repo is broken try this url:
> > http://hamspirit.pl/SQ9MDD/TTGO-T-Beam-LoRa-APRS/
> 
> Sorry, but "DOMAIN NAME: hamspirit.pl 's billing period had finished". It happened 18 jan, as I can see. Can you prolong this domain name or migrate to another and give here new link?

Use IP instead of domain name
http://83.13.245.76/SQ9MDD/TTGO-T-Beam-LoRa-APRS/

VR2XKP, 2022-02-13:
Hello

But the http://hamspirit.pl/SQ9MDD/TTGO-T-Beam-LoRa-APRS/ cannot found the latest firmware as in github.

VR2XKP, 2022-02-14:
> Hi with unknowr reasons Github locked releases.
>
> I can offer my own repo on mega.nz and you can download firmware for your
> board.
>
>
>
> https://mega.nz/folder/UZwWmSDL#IQEsBhd3fbHyc7iFEM-QpQ
>
>
>
> Have a fun :)
>
> Agee M0IGA

Dear M0IGA

Thank you for your help.

BR
VR2XKP