SQ9MDD/TTGO-T-Beam-LoRa-APRS #98: Possibility to set the frequency up to four digits behind a decimal point.

M0IGA, 2022-01-20:
In the UK we have granted a different frequency for LoRa than the rest of Europe has. It's 439.9125 but the configuration allows 3 decimal digits only. In fact, we need 4 decimal digits configuration possibility.

I tried to edit the configurator HTML code but it doesn't work. Probably it needs a deeper source code edition.

dl9sau, 2022-01-21:
Looking at lib/BG_RF95/BG_RF95.h, I see a the following resoultion:

bool BG_RF95::setFrequency(float centre)
{
    // Frf = FRF / FSTEP
    uint32_t frf = (centre * 1000000.0) / BG_RF95_FSTEP;
    spiWrite(BG_RF95_REG_06_FRF_MSB, (frf >> 16) & 0xff);
    spiWrite(BG_RF95_REG_07_FRF_MID, (frf >> 8) & 0xff);
    spiWrite(BG_RF95_REG_08_FRF_LSB, frf & 0xff);

    return true;
}

If BG_RF95_FSTEP is 
BG_RF95.h:#define BG_RF95_FSTEP  (BG_RF95_FXOSC / 524288)
BG_RF95.h:#define BG_RF95_FXOSC 32000000.0
-> 32000000.0/524288 = 61.03515625

python test:
>>> for i in range (0, 10):
...   f=int(((433.775+i*.0001)* 1000000.0) / 61.03515625) & 0xff
...   print("%f %d %f" % ((433.775+i*.0001), i, f) )
... 
433.775000 0 153.000000
433.775100 1 155.000000
433.775200 2 156.000000
433.775300 3 158.000000
433.775400 4 160.000000
433.775500 5 161.000000
433.775600 6 163.000000
433.775700 7 165.000000
433.775800 8 166.000000
433.775900 9 168.000000
-> Resolution of approx 50 Hz.

TTGO_T-Beam_LoRa_APRS.ino calls  rf95.setFrequency(lora_freq) firectly.
Argument is double lora_freq.

=> I see no limitation here.

Happy debuging

M0IGA, 2022-01-22:
Nice, I have no idea what did you have written.

dl9sau, 2022-01-22:
I leave this other redaers to explain.
My summary is: there's no frequency limitationin the rf driver, and no limitation deep inside in the tracker code. ->  Resolution of frequency should be in 50 HZ steps.

Oh, just saw what may cause the decimal length cut'ed to 3:
In file taskWebserver.cpp, there's a function definition:
  String jsonLineFromPreferenceDouble(const char *preferenceName, bool last=false){
    return String("\"") + preferenceName + "\":" + String(preferences.getDouble(preferenceName),3) + (last ?  + R"()" :  + R"(,)");
}

This always returns numbers with up to three decimals. -> 433.7755 will be displayed as 433.775. The number of decimal points is the second argument at the string function. More on this see
  https://www.arduino.cc/reference/en/language/variables/data-types/stringobject/
This function is used for displaying the variables. That means, it may be already 433.7755, but you see only 433.775.

But if I try to enter 433.7755, it does not work. This may be caused by
                        <input name="lora_freq" id="lora_freq" type="number" min="433.000" max="928.000" step="0.001" title="LoRa center frequency between 433.001 and 928.000. I.e. 433.775">
Here I'm not for 100% sure what to change (the decimal points in min/max value, the step value, or all).

dl9sau, 2022-01-25:
Only a small change to the web interface (in which resolution it accepts input values) was needed.
But keep in mind that, in my opinion, the hardware is far from so frequency stable.

M0IGA, 2022-01-27:
The half solution of the.

Edit index.html 

`<div>
                        <label for="lora_freq">Frequency [MHz]</label>
                        <input name="lora_freq" id="lora_freq" type="number" min="433.0000" max="928.0000" step="0.0005" title="LoRa center frequency between 433.001 and 928.000"> 
</div>`

and compile the project again.  These changes give a possibility to tune the freq. to .0005 but the configurator will show .015 (always rounds 5 up to 3 digits after the dot) but the frequency on the module is written correctly to 4 decimal digits.

Now the need is to force the configurator to show a correct frequency with four decimal digits.

dl9sau, 2022-01-27:
hmm. On my android device with google chrome browser, the frequency is stored on save in the web-interface and shown in the web-interface correctly.