SQ9MDD/TTGO-T-Beam-LoRa-APRS #101: Problem during build

iw9grl, 2022-08-04:
Describe the bug

After installed Visual Studio Code, loaded the package and selected the correct board ttgo-t-beam-v1.0
RUN --> build in platform IO

I received these messages :

.pio/libdeps/ttgo-t-beam-v1.0/Adafruit BusIO/Adafruit_SPIDevice.cpp: In member function 'bool Adafruit_SPIDevice::write(const uint8_t*, size_t, const uint8_t*, size_t)':
.pio/libdeps/ttgo-t-beam-v1.0/Adafruit BusIO/Adafruit_SPIDevice.cpp:335:61: error: invalid conversion from 'const uint8_t* {aka const unsigned char*}' to 'uint8_t* {aka unsigned char*}' [-fpermissive]
_spi->transferBytes(prefix_buffer, nullptr, prefix_len);
^
In file included from .pio/libdeps/ttgo-t-beam-v1.0/Adafruit BusIO/Adafruit_SPIDevice.h:9:0,
from .pio/libdeps/ttgo-t-beam-v1.0/Adafruit BusIO/Adafruit_SPIDevice.cpp:1:
C:/Users/iw9gr/.platformio/packages/framework-arduinoespressif32@3.10004.210126/libraries/SPI/src/SPI.h:73:10: note: initializing argument 1 of 'void SPIClass::transferBytes(uint8_t*, uint8_t*, uint32_t)'
void transferBytes(uint8_t * data, uint8_t * out, uint32_t size);
^
.pio/libdeps/ttgo-t-beam-v1.0/Adafruit BusIO/Adafruit_SPIDevice.cpp:338:47: error: invalid conversion from 'const uint8_t* {aka const unsigned char*}' to 'uint8_t* {aka unsigned char*}' [-fpermissive]
_spi->transferBytes(buffer, nullptr, len);
^
In file included from .pio/libdeps/ttgo-t-beam-v1.0/Adafruit BusIO/Adafruit_SPIDevice.h:9:0,
from .pio/libdeps/ttgo-t-beam-v1.0/Adafruit BusIO/Adafruit_SPIDevice.cpp:1:
C:/Users/iw9gr/.platformio/packages/framework-arduinoespressif32@3.10004.210126/libraries/SPI/src/SPI.h:73:10: note: initializing argument 1 of 'void SPIClass::transferBytes(uint8_t*, uint8_t*, uint32_t)'
void transferBytes(uint8_t * data, uint8_t * out, uint32_t size);
^
.pio/libdeps/ttgo-t-beam-v1.0/Adafruit BusIO/Adafruit_SPIDevice.cpp: In member function 'bool Adafruit_SPIDevice::write_then_read(const uint8_t*, size_t, uint8_t*, size_t, uint8_t)':
.pio/libdeps/ttgo-t-beam-v1.0/Adafruit BusIO/Adafruit_SPIDevice.cpp:429:59: error: invalid conversion from 'const uint8_t* {aka const unsigned char*}' to 'uint8_t* {aka unsigned char*}' [-fpermissive]
_spi->transferBytes(write_buffer, nullptr, write_len);
^
In file included from .pio/libdeps/ttgo-t-beam-v1.0/Adafruit BusIO/Adafruit_SPIDevice.h:9:0,
from .pio/libdeps/ttgo-t-beam-v1.0/Adafruit BusIO/Adafruit_SPIDevice.cpp:1:
C:/Users/iw9gr/.platformio/packages/framework-arduinoespressif32@3.10004.210126/libraries/SPI/src/SPI.h:73:10: note: initializing argument 1 of 'void SPIClass::transferBytes(uint8_t*, uint8_t*, uint32_t)'
void transferBytes(uint8_t * data, uint8_t * out, uint32_t size);
^
Compiling .pio\build\ttgo-t-beam-v1.0\lib35b\AXP202X_Library\axp20x.cpp.o
*** [.pio\build\ttgo-t-beam-v1.0\lib7f9\Adafruit BusIO\Adafruit_SPIDevice.cpp.o] Error 1
======================================================================================= [FAILED] Took 7.93 seconds =======================================================================================

Environment Status Duration

ttgo-t-beam-v1.0 FAILED 00:00:07.931
================================================================================== 1 failed, 0 succeeded in 00:00:07.931 ==================================================================

Screenshots

image

Desktop (please complete the following information):

OS: Win11
PlatformIO Version: last one @ Aug 4th 2022
Board name:

TTGO TBEAM 1.0
Thank you
73 de IW9GRL

[image]

gorzynsk, 2022-08-31:
Could you try to build "Default" target first? After that building only one target shall work...

ManuelFollmann, 2023-02-16:
Hello,
I got the same issue and it's caused by using espressif @3.0.0 in platformio.ini.

The change https://github.com/espressif/arduino-esp32/pull/3356 was implemented in Arduino framework 1.0.5
and this is available in platformio 3.1.0 first.

The fix is to change @ 3.0.0 to @ 3.2.0
3.1.0 causes another issue